BCN3D/BCN3DSigma-Firmware #198: Can't compile BCN3D Sigma Firmware v01-2.0.2

UBGL, 2018-09-21:
I want to compile the new firmware, but got the following error:

In file included from sketch\Hysteresis.cpp:27:0:

sketch\language.h:5:20: fatal error: macros.h: No such file or directory

 #include "macros.h"

                    ^

compilation terminated.

exit status 1

AlejandroGarcia92, 2018-09-22:
@UBGL Let me look, you will be answered next week asap.

thesec, 2018-09-22:
Same on arduino 1.8.7.

thesec, 2018-09-23:
Having moved over the 1.27 macros and after having made the HardwareSerial.h public I am now at "In file included from /var/folders/_l/fjsbcnjx489b66mnk8ph_w4m0000gn/T/arduino_build_637766/sketch/LCD_FSM.cpp:23:0:
register_codes.h:19:23: error: invalid digit "8" in octal constant
#define REGIST_CODE_8 0188".

AlejandroGarcia92, 2018-09-24:
I will take a look tomorrow. I will check what happened with the git commit as well.

AlejandroGarcia92, 2018-09-25:
@thesec I have the same issue, lets see.

AlejandroGarcia92, 2018-09-25:
Now I can compile, tell me if you can too.

thesec, 2018-09-25:
Thanks Alejandro.

Marlin_main.cpp has another include of the missing macros.h.
Removing that include, too, compilation on a fresh 1.8.7 install of arduino now ends in a 

> Marlin_main.cpp:1022:12: error: missing binary operator before token "("
>  #if ENABLED(SHOW_TEMP_ADC_VALUES)
>             ^
> Marlin_main.cpp:1047:13: error: missing binary operator before token "("
>   #if ENABLED(SHOW_TEMP_ADC_VALUES)
>              ^
> Marlin_main.cpp:1056:13: error: missing binary operator before token "("
>   #if ENABLED(SHOW_TEMP_ADC_VALUES)
>              ^
> Marlin_main.cpp:1063:13: error: missing binary operator before token "("
>   #if ENABLED(SHOW_TEMP_ADC_VALUES)
>              ^
> Marlin_main.cpp:1071:13: error: missing binary operator before token "("
>   #if ENABLED(SHOW_TEMP_ADC_VALUES)
>              ^
> /var/folders/_l/fjsbcnjx489b66mnk8ph_w4m0000gn/T/arduino_build_635937/sketch/stepper.cpp: In function 'void display_check_rx()':
> /var/folders/_l/fjsbcnjx489b66mnk8ph_w4m0000gn/T/arduino_build_635937/sketch/stepper.cpp:182:18: note: #pragma message: Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!
>   #pragma message "Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!"

The macros still seem required?

Could you drop us your mods to the HardwareSerial.h and to see what we might be missing to have it compile?

AlejandroGarcia92, 2018-09-25:
Yes, it seems that is required, I don't remember why. macros.h is linked to this direction on my computer: <C:\Users\USER\AppData\Local\Temp\VMBuilds\Marlin\mega_atmega2560\Release\macros.h> and it could be the reason about it works on my computer. I add the macros.h to the project and HardwareSerial.h.

AlejandroGarcia92, 2018-09-25:
You will find HardwareSerial.h at Marlin/BCN3D_libmods/

thesec, 2018-09-25:
Still throwing warnings

> /var/folders/_l/fjsbcnjx489b66mnk8ph_w4m0000gn/T/arduino_build_209638/sketch/stepper.cpp: In function 'void display_check_rx()':
> /var/folders/_l/fjsbcnjx489b66mnk8ph_w4m0000gn/T/arduino_build_209638/sketch/stepper.cpp:182:18: note: #pragma message: Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!
>   #pragma message "Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!"

but am back compiling now - yay! 😃

phanphan19, 2019-01-06:
Hello, i try to upload marlin 2.0.6 and i add the file "HardwareSerial.h" in c/arduino/hardaware/....../arduino but the message appear all the time like error in arduino ide 1.8.8.
For upload i choose board arduino mega 2560 and processor 2560.
Can you help me please ?

[1](https://user-images.githubusercontent.com/46418080/50736600-7fad4b80-11bf-11e9-96b9-0c6e71a6902c.PNG)

AlejandroGarcia92, 2019-01-07:
It seems that "HardwareSerial.h" is not the modified file. Did you update arduino just after adding the file?

phanphan19, 2019-01-07:
Hello, here is the procedure I am doing:
1. I download the firmware on github
2. I get the hardwareserial.h file in the zip file that I transfer to my desktop
3. I copy the file to the root file arduino c / programfiles / arduino ...... / cores / arduino.
4.I close all the tabs
5. I recover the firmware in the zip file and open it.
6. I check the firmware by putting the arduino mega card 2560 processor 2560.
7. the error appears

In your opinion where can the error come from?
Thank you again for your help, sorry for disturb.