BCN3D/BCN3DSigma-Firmware #199: Cura not connecting to printer.

Lostinthe907, 2018-09-21:
I am having this issue, i did the last update to firmware 1.2.7 and cura is up to date 2.1.0 and it is not connecting to the printer to upgrade the firmware even though it says its connected. it just sits and says connecting... 

any help to upgrade it to the new firmware not using cura or another way would be awesome

Thanks! 
[image]

Lostinthe907, 2018-09-22:
Update, 

I have reinstalled cura 2.0.2 and have tried with the same issue. i now believe it is a fault in my firmware not allowing it to connect.

Lostinthe907, 2018-09-22:
fixed!! 
Issue was the port number. When plugged into usb it was claiming COM5. renamed COM5 to COM6 and it was able to connect through Cura.

AlejandroGarcia92, 2018-09-22:
Perfect!

closed 2018-09-22 by AlejandroGarcia92