BCN3D/BCN3DSigma-Firmware #200: Something wrong with new firmware / does not load filament

omersa, 2018-09-24:
I have installed new version 01-2.0.2 on my Sigma Original upgraded to 2017. When I try to load filament a strange noise is comming from the machine and the filament is not moved. Also there is no sign of warming up. Also when the heads are moved it is much more noisier than before. Please help, as I really need ASAP.

omersa, 2018-09-24:
The same thing when trying to unload. It seems some of the steppers are hyperspeeding. When the heads move to home position they work normally, but when moving to the centre they move very fast with a lot of noise.

AlejandroGarcia92, 2018-09-24:
Have you configured properly the extruder config at Utilities/Printer Setup/Components Setup? Take into account that R19 extruder works with more steps per mm than R17.

AlejandroGarcia92, 2018-09-24:
@omersa That's a extruder issue, isn't it?

omersa, 2018-09-24:
I have selected all the right things. I can do it again. Is it enough just to change the LCD files?

omersa, 2018-09-24:
I have found a solution. At least I maneged to load the filament. There is a setting by default that the filament sensors are installed. It seems to be a bug. I selected Sigma 16/17 and this setting should be OFF by default. Will do some tests more.

AlejandroGarcia92, 2018-09-24:
Ok, If the filament sensor is on, it would work fine usually. The button filament sensor was implemented in case there was a issue on this functionality. But I didn't expect this particular issue. 

Let me know more. We are working on hotfix update for october anyway.

Thanks for the feedback.

closed 2018-09-25 by AlejandroGarcia92