BCN3D/BCN3DSigma-Firmware #201: No USB connection from Cura to Sigma R16 after upgrade to 2.0.2

SigmaFred, 2018-10-01:


AlejandroGarcia92, 2018-10-02:
Does Cura detect which FW version have you installed?

SigmaFred, 2018-10-02:
reopened - sorry for the wrong click

AlejandroGarcia92, 2018-10-03:
There are a few things that should happen and are not happening. 

- Latest version should appear if you are connected to Internet.

- "Upload custom Firmware" should work anyway.

- Just before, you loaded successfully the firmware and now not.

So, at this point, I'm not able to give you the solution but maybe we would find the problem.

**If the problem comes from CURA:**

Maybe you can try to load the firmware with another version of CURA or a program capable of doing it.

**If the problem comes from Sigma:**

Check the connection or the Port config at PC. (I don't know what to test more)

SigmaFred, 2018-10-03:
**Got it running finally!**
But with Windows machine only. With my Mac it still does not work.   
Thank you for your advice (and maybe you did a magic spell - without telling me)

Details: 
- Did a restart of Sigma (off/on)
- Shut down my Win Machine
- Started Win machine with Sigma (powered on) plugged in (USB). 
- Checked COM port (on my machine COM3) and changed default 'Baud' from 9600 to 230400.
- Started Cura (2.1.0) - both "current" and "latest version" displayed correctly.
- Then I clicked on "Automatically upgrade ..." and fortunately it worked (btw. with 'rattling' / vibrating motors on Sigma during firmware upload). 
Note: 
- In the meantime I reverted 'Baud' back to 9600 - and it is still working - so no clue if just the change made a difference (or a magic spell of Alejandro)

closed 2018-10-03 by SigmaFred