BCN3D/BCN3DSigma-Firmware #203: using heatbed logic out to switch on and off external gear?

kevinkevinkevinkevinkevinnnnnn, 2018-10-17:
hi there

not sure if this a place to ask this question but anyway lets try it:
i wanted to ask if there is any possibility to have an analog logic output of the m140 and m190 command (or whatever the heatbed gcode command is on sigma) through a pin on the mainboard to run a relais? like a hi when its activated and a low when its done or vice versa.

or would there be any kind of possibilty for this through modification of firmware or gcode? 

i hope i could explain what i want to know. im not an engineer of any kind, but know that i want a logic out or some kind of hi-low output which can be controlled by gcode

AlejandroGarcia92, 2018-10-18:
MB Led pins(24V) are working with PWM-8bit signal and the light intensity is proportional to the PWM duty cycle configured. 

But I don't what you need to do exactly.

kevinkevinkevinkevinkevinnnnnn, 2018-10-18:
hi alejandro

thanks for your reply.
i want to be able to turn on and off a relais with gcode. the relais is controlling a small heater to heat up the air inside of the printer to about 50°C(its limited by a thermostat)
is that theoretically possible with these pins. or is there any pinset on the board which can be used for communication with outside gear?
else i will just use a external timer to turn on and off. but controlling the heater with gcode would be more elegant.

best
kevin

> Am 18.10.2018 um 12:05 schrieb Alejandro Garcia <notifications@github.com>:
> 
> B Led pins(24V) are working with PWM-8bit signal a

AlejandroGarcia92, 2018-10-30:
Sorry, Led pins are not prepared to do that task(I check it again and it is no possible). There are some pins available on the mainboard but not ready to be controlled via gcode with the current FW.