BCN3D/BCN3DSigma-Firmware #204: Display Fw missing "Botton"s

thesec, 2018-10-19:
Workshop4 IDE does not open the .4dgenie project file for a ton of 'Botton's claimed missing, e.g.
[image]

thesec, 2018-10-19:
Updating the IDE to the latest - 4.5.0.17b - ... just stalls the IDE :/

thesec, 2018-10-21:
Finally stumbled across the "Read.txt" in "ExtraFiles" and got to find the location described by "*\4D Labs\4D Workshop 4 IDE\SysImages\Buttons".

I suggest to include instead the absolute path as Windows users rarely install on other paths and only the drive letter should change if they do.
Moving the ReadMe next to the Genie file will help others to find it more quickly.

AlejandroGarcia92, 2018-10-22:
Ok, I will take into account your suggestions.

closed 2018-10-29 by AlejandroGarcia92