BCN3D/BCN3DSigma-Firmware #209: Compilation Settings Display FW

thesec, 2018-10-27:
Which Project/Destination settings in the Workshop 4 IDE are the correct ones to compile the same as provided within the repository?
"Run RAM", "Run Flash" or "uSD" and which Bank?
Which do you use to bootstrap the displays for R19's?

thesec, 2018-11-04:
Just tried:
"No Errors, code size = 0 bytes
Starting Script LOADSIGMA.4dscript
Ending Script LOADSIGMA.4dscript"
is what happens without anything loaded to the display, at least it does not reset as I'd expect it to? The com port is the one reported by "mode" and in the IDE.
Is there anything else to be set, configured, ...?

AlejandroGarcia92, 2018-11-05:
Don't open LOADSIGMA.4dscript, use the Temp4D.4dg.
 You will see something like that: 

I pressed compile

[image]

[image]

[image]

thesec, 2018-11-07:
Thanks again, hope to get to give it a try next weekend ...

thesec, 2018-11-15:
Yay - that worked! Thanks @AlejandroGarcia92 👍

closed 2018-11-15 by thesec