BCN3D/BCN3DSigma-Firmware #210: [Bug] R16 Unable to print from SD

pvnct, 2018-11-12:
After upgrading my R16 to the latest firmware 2.0.4 (and the screen sd card of course) i am unable to print from SD card: 

All files appears in the browser but when i hit print it reload the browser and don't launch the print. 

I have tried  with different  SD cards, different codes from BCN3D cura or S3D , same result

AlejandroGarcia92, 2018-11-13:
I discard mainboard FW cause of course and I think that something is failing during lcd fw upgrade. Historically there were some issues around the sd card but I'm not sure what is happening right now without more extra feedback.

AlejandroGarcia92, 2018-11-14:
Any way, can you tell us which sd card do you have placed on the LCD? Because we had some problems with the sd card that are not industrial, and those are what we are providing right now on all our products(Printers and Upgrade kid R19).

If you can take a picture, I will be glad.

pvnct, 2018-11-16:
The Screen SD has been provided by BCN3D - You / your team already replaced the original one i got with the  machine. 
I remember from the last failed update that it is a Toshiba (red and grey-white) but i will take a pic as soon as i return to my office. 
Also i have to say that i didn't try to format the screen sd  to FAT16 (if i remember well) when i tried to update to 2.0.4 
Perhaps it can helps

But what is weird is that the only thing not working were launch print. Alternatively i have noticed a faster file browsing and maybe also a faster start  - new features like extruder setup (bondtech) - or even the led dimming

pvnct, 2018-11-19:
[image]

AlejandroGarcia92, 2018-11-19:
It is not a good sd card. Can you try to format the sd anyway?

pvnct, 2018-11-19:
FAT16 ? as previous fw version ?

AlejandroGarcia92, 2018-11-19:
Yes FAT16 for the LCD

closed 2020-07-21 by AlejandroGarcia92