BCN3D/BCN3DSigma-Firmware #213: Sigma R16 - E3D V6, 1.75 BMG

lkvapil, 2018-12-27:
Hi,

I upgraded my Sigma R16 with E3D V6 hotends (EmvioEngineering parts) and Bondtech BGM extruders (1.75mm DIA). I downloaded the latest FW, edited thermistor types, changed extruder steps and DEFAULT_NOMINAL_FILAMENT_DIA to 1.75. I sliced a test cube in Simplify3D and it printed well with left hotend, but when I do full calibration via the display, hotends are in different positions at X axis. (they are off) 
Which parameter I have to change in configuration.h file ? 
(I have at both X motors 16th pulley) 

I'm looking forward to response
Thank you 

Lukas

AlejandroGarcia92, 2019-01-02:
#define X_SIGMA_PROBE_OFFSET_FROM_EXTRUDER  17.05//20
#define Y_SIGMA_PROBE_OFFSET_FROM_EXTRUDER	22.2
#define Z_SIGMA_PROBE_OFFSET_FROM_EXTRUDER  2.8//2.80//3.4 //It is negative, it is compensated
		
#define X_SIGMA_SECOND_PROBE_OFFSET_FROM_EXTRUDER	-13.4
#define Y_SIGMA_SECOND_PROBE_OFFSET_FROM_EXTRUDER	22.2
#define Z_SIGMA_SECOND_PROBE_OFFSET_FROM_EXTRUDER	2.8//2.90

lkvapil, 2019-01-02:
Hi,

and steps for stepper motors X0 & X1 and E0 & E1 are in Firmware same ?

#define DEFAULT_AXIS_STEPS_PER_UNIT {100,80,1600,400} - both X0 & X1 motors
100 steps/mm and both E0 & E1 400 steps/mm ?

Thank you
Lukas

AlejandroGarcia92, 2019-01-03:
Yes

closed 2019-02-04 by AlejandroGarcia92

lkvapil, 2019-08-27:
> I have the same in mind but also adding a Zesty Nimble to get remote
> direct drive extruder(s). May I ask how to compile the Arduino files?
> (Which libraries to use & board to select?)

add me through messenger if you can : Lukas Kvapil

I would like to help

Lukas

fjohannes, 2019-08-27:
I have the same in mind but also adding a Zesty Nimble to get remote direct drive extruder(s). May I ask how to compile the Arduino files? (Which libraries to use & board to select?)

fjohannes, 2019-08-29:
which kind of messenger you refer to?
there is none in github, isn't it?

lkvapil, 2019-09-02:
how did it went up?

fjohannes, 2019-09-25:
Hi Lukas. Sorry, I still didn't see a way to contact you directly here.

Anyway I made some progress: I was able to compile the firmware in Arduino 1.8.5 with replacing the HardwareSerial.h in the Arduino directory beforehand. It sucessfuly compiles and I can create .hex files (with and without bootloader) to upload via BCNcura to the machine (I upload the file without bootloader).

Two remaining issues:
1. During compiling I (in 75% of the cases) get the following message:
`C:\Users\Johannes\AppData\Local\Temp\arduino_build_943688\sketch\stepper.cpp: In function 'void display_check_rx()':
C:\Users\Johannes\AppData\Local\Temp\arduino_build_943688\sketch\stepper.cpp:182:18: note: #pragma message: Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!
  #pragma message "Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!"`

I don't know what to change here. In the code of HardwareSerial.h there's a line 
`class HardwareSerial : public Stream` which sounds ok.
Then I assumed the file permissions are meant and tried: 1. log in as admin & compile - same issue, 2. run Arduino.exe as administrator & compiled - same issue, 3. changed the permissions of the file to full access for all users but no success (using Win7pro64bit)

2. The firmware is sucessfully uploaded and I can also verify by displaying an altered version number (I added a suffix to the version 01-2.0.8) on the Sigma LCD.
Anyway, the changes I (need to) make in the DEFAULT_AXIS_STEPS_PER_UNIT don't seem to influence the machine behaviour. When I control the machine over Cura and purge E with 10mm e.g., it moves the same distance no matter how high I change the value for E in DEFAULT_AXIS_STEPS_PER_UNIT in the firmware...

fjohannes, 2019-09-25:
I just called the EEPROM settings with M503 in Pronterface and for Steps per unit it gives values, which I found in configuration.h are defined for the MEGATRONICS_V3 board option...?! Anyway, I have defined `#ifndef MOTHERBOARD
	#define MOTHERBOARD BCN3D_BOARD
#endif`  and changed my values in the correct line for the BCN3D_BOARD

lkvapil, 2019-09-25:
+420739157590
my whatsapp, message me dude