BCN3D/BCN3DSigma-Firmware #216: "Serial print job" screen should only show when printer is actually printing

JanneMantyharju, 2019-03-10:
I'm using Sigma in combination with Octoprint. The newest firmware introduced the "Serial Print Job" screen which is visible when printer is printing from usb. The only problem is that this screen shows up as soon as the external device opens connection and after that it's not possible to use the printers screen to command filament change etc.

I'd like to have this changed that the screen is visible only when the gcode is streaming from external device

AlejandroGarcia92, 2019-03-11:
Hi @JanneMantyharju,

Configure Octoprint with the following mods:

* Go to Octoprint Setting->GCODE Scripts.

* Insert the following code at "After print job is cancelled":

```
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0
```
* Finally, insert the following code at "After connection to printer is established":

```
;Remove printing by serial message (debug)
M84
```

Tell me if it works for you.

JanneMantyharju, 2019-03-15:
Thanks, this seems to work! It would be great to document this somewhere

AlejandroGarcia92, 2019-03-18:
Wiki could be a good place.

closed 2019-03-18 by AlejandroGarcia92