BCN3D/BCN3DSigma-Firmware #217: LCD Image rotated after the LCD firmware update SD files 2.0.2

mm2700, 2019-04-02:
Hello,
I updated my BCN3D Sigma R17 via Cura to 2.0.6. In order to install the matching LCD Firmware, I disassembled the printer to reach the SD-Card of the LCD-Screen.
I formatted the in Windows with the FAT-Filesystem and 64KB Cluster Size. After that I copied the files for the R17-R19 to the SD-Card. Everything is working but the screen is 180 Degrees rotated. 

After that I formatted the SD-Card again and tried the old backup of the SD-Card. The Rotation seems fine. After that I redownloaded the 2.0.2 files for R17-R19 but unfortunately without success. The Image is still upside down.

AlejandroGarcia92, 2019-04-03:
Technically, the R16 files are not reversed and R17-R19 are reversed. I have no more ideas behind it.

mm2700, 2019-04-03:
> Technically, the R16 files are not reversed and R17-R19 are reversed. I have no more ideas behind it.

Thank you for your comment, I also tried the 1.2.2 LCD Files and they are also reversed. Can I try to use the R16 files with the R17?

AlejandroGarcia92, 2019-04-03:
Yes, try it.

mm2700, 2019-04-04:
Thank you, it works with the R16 with no issues so far. Is the difference between the R16 and R17-19 files just the LCD orientation?

AlejandroGarcia92, 2019-04-04:
Yes, the files itself are the same but reversed.

closed 2019-04-04 by AlejandroGarcia92