BCN3D/BCN3DSigma-Firmware #218: Stuck in Calibration

thesec, 2019-08-01:
Leaving the Sigma in the left Z-calibration I got distracted a few minutes ... and found the printer having printed the calibration mesh but also having automatically allowed himself to return to the main menu.
Restarting the calibration it starts again with the left Z-calibration but also returns to the main menu right afterwards. A cold restart - turning power off and back on - of the printer did not help.

How to get out of this loop?

Firmware in use is 2.07.

thesec, 2019-08-01:
Keeps doing the same after an update to 2.08 ... and ... also after a 'factory reset' using M502 ... :/

The prompt to select one of the five printed lines shows up for a blink and is immediately replaced by the main menu.

thesec, 2019-08-01:
Keeps showing up after bootstrapping the board ... wtf :#

fjohannes, 2019-08-01:
Had the same issue with v01-208 firmware twice, but luckily after cold restart it worked again.

AlejandroGarcia92, 2019-08-02:
Hi, you did a good description of the issue but it sounds very rare. Maybe, the element that it's not working is the LCD screen. That's the only idea that I have in mind at the moment. So, SD card format could be enough but I'm not sure.

thesec, 2019-08-02:
Thanks Alejandro, already been under way in that direction having checked the SD to sit properly and the cables to do so, too. Will add a reformatting ...

closed 2020-07-21 by AlejandroGarcia92