BCN3D/BCN3DSigma-Firmware #219: Smartpurge M800 does not work

mfreda, 2019-08-27:
Hi, I tested M800 command: M800 F48 S0.0003 E15.25 P3
Smartpurge function return always (after tool switch):

READ: Purge distance
READ: 3.00
READ: Time inactive
READ: 0

Analysing the code (version 2.0.8), I noticed that variable time_inactive_extruder[0|1] is never incremented.

Best Regards.

Mario Freda.

AlejandroGarcia92, 2019-08-28:
Let me check it.

AlejandroGarcia92, 2019-08-29:
I've seen that this feature is implemented only for SD printing.

mfreda, 2019-08-29:
Ok, where the variabile time_inactive_extruder[0|1] is incremented in source code Marlin_main.cpp?

In the old versions was incremented in display loop. In the new version I cannot found where is incremented.

Thank you very much.

Mario Freda.

UBGL, 2019-09-08:
@AlejandroGarcia92

Is it possible to make it in serial printing available?

AlejandroGarcia92, 2019-09-09:
Yes, it is.

closed 2020-07-21 by AlejandroGarcia92