BCN3D/BCN3DSigma-Firmware #221: No Filament runout detection during a serial print job

UBGL, 2019-09-08:
I notice that the printer didn't stop/wait if the filament is running out when print job is comming over serial connection. Didnt notice any infos in the print log. Using Repetier-Server.

AlejandroGarcia92, 2019-09-12:
Filament runout detection is not supported for serial prints. In case of implementation, it would be supported by BCN3D Cura.

conraddewet, 2020-05-20:
It is sad that this is not enabled in the firmware. The Sigma/Sigmax works great with Octoprint, only thing that is missing is the runout detection. With M600 not supported, we can not even add external sensors.
How do we get support for M600 and M412 ? If this works then my Sigmax will work great as an online printer - even my $300 printers supports this.

AlejandroGarcia92, 2020-07-21:
#226

closed 2020-07-21 by AlejandroGarcia92