BCN3D/BCN3DSigma-Firmware #225: Saving the print to resume the next day isnt working

DLH1997, 2019-10-08:
Hi ,

I'm using a BCND3D and when I leave work I have to turn off the printer , so I save the print and the next day when I turn on the printer and resume the print it doesn't print precisely on top of the last layer but slightly beside it ( the whole print shifts about a millimetre forward in the z axis). 

Ive updated the firm ware 2 weeks ago so I don't know if I need to re-update it? Would you have any idea what would be causing this problem?

AlejandroGarcia92, 2019-10-14:
This features doesn't work properly because there are a few factors that are affecting to not succeed. For instance, the repositioning itself and the material contraction after cooldown.

DLH1997, 2019-10-17:
Ah so it's the feature itself that doesn't work full stop ? So you have this problem too when you use this feature?