BCN3D/BCN3DSigma-Firmware #226: Continue Development? Any Progress? Any Soon?

thesec, 2020-03-26:
Hi the repo has ceased active development 9 months ago ... 
Do you plan to give birth any soon?
Is there any development going on?

Just a short ping of life would be nice :)

AlejandroGarcia92, 2020-03-26:
Hi @thesec,
The development continues with Epsilon and new gen. We are developing Marlin there: https://github.com/BCN3D/Marlin, Right now, Marlin is an small part of firmware development.

thesec, 2020-03-26:
Thanks for the swift heads-up @AlejandroGarcia92 ,

the new repo being a recent fork and already inluding specifics to the Epsilon ... will it be finished first for it and then adapted for Sigma (X)? Is it planned to be adapted or downwards compatible?

AlejandroGarcia92, 2020-07-21:
It's not planned to continue the Sigma/x development and the downwards compatibility neither.

closed 2020-07-21 by AlejandroGarcia92

AlejandroGarcia92, 2020-07-21:
Sorry about my previous message, it was quite inaccurate and I'm aware that it has raised some concerns regarding the continuation of the Sigma line, so I have decided to delete it.

We are going to continue working on constantly improving our products as we have done so far and, of course, we are working on new developments at all levels.
The Sigma line has been our main focus for a very long time and, while now we also count on the BCN3D Epsilon and we are splitting our efforts between a bigger portfolio, we are not going to abandon our Sigma/x users, nor are we pronouncing the Sigma/x dead. I can assure you that the Sigma line has a long future ahead!

thesec, 2020-07-21:
"splitting our efforts" is very different from stalling development completely on that line for over a year now.

I think you have to show actual efforts now to gain a few credits again.

AlejandroGarcia92, 2020-07-21:
It's an evidence that we are more focused on the new gen and It's true that this repository has not been updated for a year. In the short term we will not launch an update, however, it doesn't implies that we are not going to launch an update anymore. I mean that we are still listening the community and if we can launch an updated that improves the printer we will do it of course.

thesec, 2020-07-22:
Well, the community is listening, too, and sees lots of issues still pending to be improved if not fixed - not only in firmware, other ends are even worse ... 

So not launching any update for over a year with improvements all around ... just do it.
All other just does not sound very ambitious.