BCN3D/BCN3DSigma-Firmware #235: Random "unload filament" during print

QuInsight, 2021-05-19:
Sometimes a print will unexpectedly go into "Unload Filament" sequence even though I have not implemented color or filament changes in the print. On a recent 18-hour print the problem recurred every 5 minutes or so during the last 2 hours. Problem has occurred on several print files at random over past several months. Unable to find current bug report or fix that matches my problem.

Prusa Slicer 2.3.0
i3 MK3

freddotu, 2021-05-19:
I can see how that's a disturbing development. I see you are using Prusa Slicer. I have considered to use that slicer for my Sigma, but have not found a profile that supports dual extrusion with duplication and mirroring. Is this something you've found online or developed yourself? Does the profile you use support duplication and mirroring as well as dual material/dual colors?

If one knew the g-code that triggers a filament change sequence, it could be discovered how that is injected into the code when slicing.

QuInsight, 2021-05-20:
Here is a g-code that gave me the unexpected filament unload. It's a roughly 20 minute print with the error occurring at around minute 17. One file in zip.

[Unexpected Filament Change.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/6512369/Unexpected.Filament.Change.zip)

QuInsight, 2021-05-20:
Gosh darn...I see I am in the wrong place. New to git hub.

closed 2021-05-20 by QuInsight